FlyGoat/RyzenAdj #315: Disregarded power limit on Ryzen 7535U

tova-cup, 2024-09-21:
Hello,

I am running Debian 12 on my Thinkpad laptop with a Ryzen 7355U and I noticed that despite setting power draw limits on RyzenAdj there's no effect on the real power consumption.

Practically, I did run 
`# ryzenadj --stapm-limit=2000 --fast-limit=5000 --slow-limit=2000 --apu-slow-limit=2000`

And yet, the monitoring returns

> CPU Family: Rembrandt
> SMU BIOS Interface Version: 18
> Version: v0.16.0 
> PM Table Version: 450005
> |        Name         |   Value   |     Parameter      |
> |---------------------|-----------|--------------------|
> | STAPM LIMIT         |     5.000 | stapm-limit        |
> | STAPM VALUE         |     2.727 |                    |
> | PPT LIMIT FAST      |     5.000 | fast-limit         |
> | PPT VALUE FAST      |     2.933 |                    |
> | PPT LIMIT SLOW      |     2.000 | slow-limit         |
> | PPT VALUE SLOW      |     2.748 |                    |
> | StapmTimeConst      |       nan | stapm-time         |
> | SlowPPTTimeConst    |       nan | slow-time          |
> | PPT LIMIT APU       |     2.000 | apu-slow-limit     |
> | PPT VALUE APU       |     0.000 |                    |
> | TDC LIMIT VDD       |    50.000 | vrm-current        |
> | TDC VALUE VDD       |     1.694 |                    |
> | TDC LIMIT SOC       |    13.000 | vrmsoc-current     |
> | TDC VALUE SOC       |     0.781 |                    |
> | EDC LIMIT VDD       |   105.000 | vrmmax-current     |
> | EDC VALUE VDD       |     7.505 |                    |
> | EDC LIMIT SOC       |    17.000 | vrmsocmax-current  |
> | EDC VALUE SOC       |     0.040 |                    |
> | THM LIMIT CORE      |    70.000 | tctl-temp          |
> | THM VALUE CORE      |    39.369 |                    |
> | STT LIMIT APU       |    43.000 | apu-skin-temp      |
> | STT VALUE APU       |    27.520 |                    |
> | STT LIMIT dGPU      |     0.000 | dgpu-skin-temp     |
> | STT VALUE dGPU      |     0.000 |                    |
> | CCLK Boost SETPOINT |       nan | power-saving /     |
> | CCLK BUSY VALUE     |       nan | max-performance    |

Please note the breaching of **slow-limit**.
Moreover, judging from the ryzenAdj readings, I would expect the actual power draw to be around 2.7W.

However powertop returns a consumption around 6W :

> The battery reports a discharge rate of 6.12 W

FlyGoat, 2024-09-21:
Battery discharge rate do include other system components such as monitor and memory, while ryzenadj can only read CPU power consumption, so that makes sense to me.

I think 2000 might be a bit too low so SMU refused to apply such limit.

tova-cup, 2024-09-21:
Thank you for the clarification.
I'll close the issue then.

By the way, is the TDP limit set by RyzenAdj only for the CPU or the combined couple CPU+iGPU ?

closed 2024-09-23 by tova-cup