FlyGoat/RyzenAdj #316: Please increase support for ryzen 5 5500u (Lucienne)

abhishekcs0, 2024-09-30:
TDP settings are working but can't increase gfx max clock, It works with AATU on windows with this apu but I am on linux.

Using preset: --stapm-limit=30000 --fast-limit=34000 --slow-limit=32000 --slow-time=128 --stapm-time=64 --tctl-temp=85 --vrm-current=180000 --vrmsoc-current=180000 --vrmmax-current=180000 --vrmsocmax-current=180000 --max-gfxclk=1850 --apu-skin-temp=85
Auto reapply: Disabled
--------------- RyzenAdj Log ---------------
Sucessfully set stapm_limit to 30000
Sucessfully set fast_limit to 34000
Sucessfully set slow_limit to 32000
Sucessfully set slow_time to 128
Sucessfully set stapm_time to 64
Sucessfully set tctl_temp to 85
Sucessfully set vrm_current to 180000
Sucessfully set vrmsoc_current to 180000
Sucessfully set vrmmax_current to 180000
Sucessfully set vrmsocmax_current to 180000
set_max_gfxclk_freq is not supported on this family
Sucessfully set apu_skin_temp_limit to 85

FlyGoat, 2024-09-30:
Try: https://github.com/FlyGoat/RyzenAdj/commit/a89d59c1b7f99b527e3b14a80a0b4edb47de0e42

abhishekcs0, 2024-10-01:
Sorry I am complete noob what to do with this, or are you trying to say you fixed this issue and I should download latest version of ryzenadj ?

FlyGoat, 2024-10-01:
> Sorry I am complete noob what to do with this, or are you trying to say you fixed this issue and I should download latest version of ryzenadj ?

If you are on Windows, try this build: https://github.com/FlyGoat/RyzenAdj/actions/runs/11105608681/artifacts/1995231355

If you are on Linux, clone the master branch and compile your own build.

Please report if it works, Thanks.

abhishekcs0, 2024-10-02:
Still showing

`set_max_gfxclk_freq` is not supported on this family`

abhishekcs0, 2024-10-02:
Now able to set it 
`Sucessfully set max_gfxclk_freq to 1900'

but it's not working clock is not going beyond default 1800

kimhuilim, 2024-10-02:
Hi, can confirm it does work to when setting it. But doesn't get applied during benchmarks. It just says successful.

FlyGoat, 2024-10-03:
Thanks for the report, `gfxclk_freq` can't go beyond CPU's fused limitation.

From AMD's product spec for R5 5500U (https://www.amd.com/en/support/downloads/drivers.html/processors/ryzen/ryzen-5000-series/amd-ryzen-5-5500u.html) the graphics frequency is 1800MHz for this chip.

There is nothing more we can do with it.

closed 2024-10-03 by FlyGoat