FlyGoat/RyzenAdj #319: build instructions are not robust enough

diggit, 2024-10-27:
Hi,
build instructions in readme assume the default `cmake` build backend is `make` which may not be the case (in my case it's `Ninja` which is faster than `make`).

Original
``` sh
cd RyzenAdj
rm -r win32
mkdir build && cd build
cmake -DCMAKE_BUILD_TYPE=Release ..
make
#...
```

Suggested
``` sh
cd RyzenAdj
rm -r win32
cmake -b build -s . -DCMAKE_BUILD_TYPE=Release
cmake --build build
cd build
#...
```

What do you think?

Ken-g6, 2024-12-07:
I decided to try this on Mint 22.  At the first cmake I got:

```
CMake Warning:
  Ignoring extra path from command line:

   "[my full path to]/RyzenAdj/build"

CMake Error: Unknown argument -s
CMake Error: Run 'cmake --help' for all supported options.
```

diggit, 2024-12-07:
@Ken-g6 sorry for that, I wrote this issue in hurry and forgot the correct case of argumente (upper case), fixed it now.