FlyGoat/RyzenAdj #320: Error/Warning message improvement

Alex-V69, 2024-10-28:
I am trying to reduce the tctl_temp value of my CPU. 

After I finished editing the configuration section of readjustService.ps1 and tried to run the script, I get this warning:

[image]

I tried to set a user environment variable with this name ("tctl_temp") but still get the same warning.

Can somebody please point me in the right direction where I can learn more about this "profile" thing?

Falcosc, 2024-10-29:
The script is only needed if values are changing and is a template for developers.
If the comments in the script are not enough to understand the configuration, then it's not advisable to proceed with it and better go for UI solutions of automation.

Alex-V69, 2024-10-30:
Hello. Thank you for the advice. I still want to understand what this "profile" is but will use a Python based option if I don't find that out.

> "reduces the power allowance to the lowest possible value, making the system completely freezing"

I use Universal x86 Tuning Utility at the moment but want to find a more robust, controllable, and lightweight solution (in this order).
When I set "Temperature Limit" via "APU Temperature Tuning" I haven't seen the behavior you described even when I played resource intensive games (neither the frame rate nor the CPU temp suggested that – the temperature just reached a plateau and there was no sudden and drastic frame rate drop).

P.S. It is an HP ProBook x360 435 G8 with an AMD Ryzen 7 PRO 5850U (Cezanne) CPU

Falcosc, 2024-10-30:
On my end hitting reduced max temps triggers prochot and reduce clocl to 500mhz

Alex-V69, 2024-10-31:
I assume thermal protection behavior depends on a CPU architecture and its manufacturer.

In my case, it's just throttling:

[image]

I get the same result when I use a modified version of readjust.py

```python

...

set_tctl_temp = 60

...

# define ctype mappings for types which can not be mapped automatically

...

lib.get_tctl_temp.restype = c_float
lib.get_tctl_temp.argtypes = [c_void_p]

...

print(f"Monitor if tctl_temp is not {set_tctl_temp}")
while True:
    lib.refresh_table(ry)
    limit = round(lib.get_tctl_temp(ry))
    print(f"Current limit is: {limit}")
    if limit != set_tctl_temp:
        print("reapply limits, because old limit was {:d}".format(limit))
        adjust("tctl_temp", set_tctl_temp)
    time.sleep(3)

```