FlyGoat/RyzenAdj #321: Cezanne / 5600u / HP 445 G8

mightatnight, 2024-11-18:
I'm trying to bypass the limits now, the issue is when i'm setting the 30w limit, it handles it for a few seconds,then drops to the 25w,and after that to 15w. I'm using the uxtu now,because it's much more simplier for me, and for me helps changing preset to the eco,and back to the custom(reapplying the same profile is not working) I've read the "Cezanne Support" issue but i didn't understand what are you talking about, used ryzenadj timer,it's working,but EasyAnticheat reacts on it(as i know it's because of getting table)

mightatnight, 2024-11-18:
Sorry if my question looks dumb, i'm also a developer,but this is not my specialization at all :(

DaniloJacques, 2025-11-24:
You need to set #resetSTAPMUsage to TRUE.
Just look for it on readjustService.ps1:
$resetSTAPMUsage = $true