FlyGoat/RyzenAdj #333: Support for Stirx Halo

kl4ng3r, 2025-03-17:
Thank you for your great work on this project.
Would you be able to update it to support Strix Halo APU's ? (AMD Ryzen Ai Max+ 395)

FlyGoat, 2025-03-17:
This is planned, I already gathered some information and will publish a test version once I get some time.

kl4ng3r, 2025-03-22:
Thank you very much for your work

geeksville, 2025-04-20:
I think probably fixed in https://github.com/FlyGoat/RyzenAdj/pull/334

FlyGoat, 2025-04-20:
Nice! Thanks for your work!

geeksville, 2025-04-21:
thank you, you did the hardest part! ;-)

Akcyn, 2025-07-24:
Will undervolt feature be added in next updates? AI Max 395+ currently have some apps with it, it will be awesome if you add this feature.

geeksville, 2025-07-24:
@Akcyn it was already added.  It is in the 0.17 release builds (last May)

wa268004, 2025-09-29:
Is setting the GPU clock not supported on Strix Halo? It doesn't seem to work. --max-gfxclk --min-gfxclk