FlyGoat/RyzenAdj #340: Possible missing break statements.

AngelouDi, 2025-05-01:
Going through the code on api.c I noticed that on many places the break statement is missing from switch.

e.g.

```c
EXP float CALL get_tctl_temp(ryzen_access ry) {
	switch (ry->table_ver)
	{
	case 0x001E0001:
	case 0x001E0002:
	case 0x001E0003:
	case 0x001E0004:
	case 0x001E0005:
	case 0x001E000A:
	case 0x001E0101:
	case 0x0064020c: // StrixHalo tested
		_read_float_value(0x58); //use core1 because core0 is not reported on dual core cpus
	case 0x00370000:
	case 0x00370001:
	case 0x00370002:
	case 0x00370003:
	case 0x00370004:
	case 0x00370005:
	case 0x003F0000:
	case 0x00400001:
	case 0x00400002:
	case 0x00400003:
	case 0x00400004:
	case 0x00400005:
	case 0x00450004:
	case 0x00450005:
	case 0x004C0006:
	case 0x004C0007:
	case 0x004C0008:
		_read_float_value(0x40);
	default:
		break;
	}
	return NAN;
}
```

Is this intentional?

Falcosc, 2025-05-01:
Did you mean switch fall through or the missing break after read float?

Fall through is one of the points why switch statements can be dangerous and be marked by many code quality rules.
It actually looks like this in compiled form. A break is just a jump instruction and without it will just execute the next block, which can be confusing.

The _read_float_value macro however does input a return there. It's not a function call on stack, it's lines of code which get inlined into this position

AngelouDi, 2025-05-01:
Yes, sorry for opening this silly issue.