FlyGoat/RyzenAdj #350: Phoenix Point, failed to get pm_table

xiota, 2025-06-02:
Current HEAD 9b8f21ecf0a580d5a1f19811bf6eaacb3dbfe110 (0.17.0.r2) is affected.  0.16.0 worked as expected.  Using Arch Linux.
```
$ sudo ryzenadj -i
CPU Family: Phoenix Point
SMU BIOS Interface Version: 14
Version: v0.16.0
failed to get pm_table from /dev/mem
refresh_table failed
Unable to init power metric table: -5, this does not affect adjustments because it is only needed for monitoring.
```

First bad commit, from git bisect, d22299f63609d86e8320b6c1d835195560782c46.
```
$ git bisect log
git bisect start
# status: waiting for both good and bad commits
# good: [4e278706f6fc7b56fe26d72927da62de5ace05be] ci: Bump actions/upload-artifact
git bisect good 4e278706f6fc7b56fe26d72927da62de5ace05be
# status: waiting for bad commit, 1 good commit known
# bad: [67aa960e71bf4cdd140b47d42c0c62c4cded68d1] Bump to 0.17.0
git bisect bad 67aa960e71bf4cdd140b47d42c0c62c4cded68d1
# good: [a07bd6761c3b0ab4b42466aac2e59fbf91a153bd] Add support for AMD Ryzen 7840U
git bisect good a07bd6761c3b0ab4b42466aac2e59fbf91a153bd
# bad: [92fc2d7c44bcea639288406da7b8bf4ebc94f436] krackan: Allow setting limits and time
git bisect bad 92fc2d7c44bcea639288406da7b8bf4ebc94f436
# bad: [d22299f63609d86e8320b6c1d835195560782c46] Remove old, incomplete ryzen_smu backend
git bisect bad d22299f63609d86e8320b6c1d835195560782c46
# good: [8425bb6c72411b0876917c10b17cca0427a3a6ea] Linux: Add ryzen_smu kernel module backend
git bisect good 8425bb6c72411b0876917c10b17cca0427a3a6ea
# first bad commit: [d22299f63609d86e8320b6c1d835195560782c46] Remove old, incomplete ryzen_smu backend
```

xiota, 2025-06-03:
Built and installed [amkillam/ryzen_smu](https://github.com/amkillam/ryzen_smu).  Basically same error:
```
detected compatible ryzen_smu kernel module
CPU Family: Phoenix Point
SMU BIOS Interface Version: 14
Version: v0.17.0
refresh_table failed
Unable to init power metric table: -5, this does not affect adjustments because it is only needed for monitoring.
```

donatin48, 2025-06-04:
I have the same issue with a Hawk Point cpu

kylon, 2025-06-04:
@RPochyly strix is fixed in #353 

phoenix and hawk, have no idea, it was not working before because the table is not supported in ryzenadj, it probably didn't print any error or you missed it, but ryzen_smu now shows inconsistencies

to confirm, paste debug log

xiota, 2025-06-04:
> your table version is not supported in ryzenadj

As described earlier, that was a regression in d22299f63609d86e8320b6c1d835195560782c46.  Regardless, now works as expected for Phoenix Point with 98d5815 and https://github.com/amkillam/ryzen_smu/commit/9f9569f889935f7c7294cc32c1467e5a4081701a.

References to ryzen_smu versions are confusing because there are no releases or tags.  As noted by another user, versions in ryzen_smu Makefile are inconsistent.

```
sudo ryzenadj -i
detected compatible ryzen_smu kernel module
CPU Family: Phoenix Point
SMU BIOS Interface Version: 14
Version: v0.17.0
PM Table Version: 4c0008
|        Name         |   Value   |     Parameter      |
|---------------------|-----------|--------------------|
| STAPM LIMIT         |    40.000 | stapm-limit        |
| STAPM VALUE         |     5.428 |                    |
| PPT LIMIT FAST      |    45.000 | fast-limit         |
| PPT VALUE FAST      |     5.378 |                    |
| PPT LIMIT SLOW      |    40.000 | slow-limit         |
| PPT VALUE SLOW      |     5.380 |                    |
| StapmTimeConst      |    89.588 | stapm-time         |
| SlowPPTTimeConst    |    59.217 | slow-time          |
| PPT LIMIT APU       |    54.000 | apu-slow-limit     |
| PPT VALUE APU       |       nan |                    |
| TDC LIMIT VDD       |    70.000 | vrm-current        |
| TDC VALUE VDD       |     1.641 |                    |
| TDC LIMIT SOC       |    18.000 | vrmsoc-current     |
| TDC VALUE SOC       |     1.098 |                    |
| EDC LIMIT VDD       |   140.000 | vrmmax-current     |
| EDC VALUE VDD       |    20.485 |                    |
| EDC LIMIT SOC       |    26.000 | vrmsocmax-current  |
| EDC VALUE SOC       |     3.050 |                    |
| THM LIMIT CORE      |    92.001 | tctl-temp          |
| THM VALUE CORE      |    37.372 |                    |
| STT LIMIT APU       |     0.000 | apu-skin-temp      |
| STT VALUE APU       |     0.000 |                    |
| STT LIMIT dGPU      |     0.000 | dgpu-skin-temp     |
| STT VALUE dGPU      |     0.000 |                    |
| CCLK Boost SETPOINT |       nan | power-saving /     |
| CCLK BUSY VALUE     |       nan | max-performance    |
```

kylon, 2025-06-04:
that specific commit did nothing to break it, you got an error because /dev/mem is restricted and in that commit ryzen_smu support is removed

the error was a table size mismatch, like strix, if it works now, it means ryzenadj had a mismatch/wrong value? till now

i already send a pr to update makefile https://github.com/amkillam/ryzen_smu/pull/10

@FlyGoat can be closed

xiota, 2025-06-04:
Removed support ⇒ previously working functionality no longer works ⇒ regression

This should not be closed until others confirm working for hawk point and strix point.

kylon, 2025-06-04:
the missing strix value in the log is now merged
yea, we can wait hawk confirmation

i don't know why it worked before, according to your output, just luck somehow, ryzenadj has a broken check, so no error, and your table was missing in ryzenadj, added in my recent commit, but present in smu module
ryzenadj check is still broken, but smu is now checked after 8425bb6c72411b0876917c10b17cca0427a3a6ea

donatin48, 2025-06-13:
It works thanks ! but the version of ryzen_smu from the aur (https://gitlab.com/leogx9r/ryzen_smu) don't work i update with this version (https://github.com/amkillam/ryzen_smu)

❯ sudo ryzenadj -i
detected compatible ryzen_smu kernel module
CPU Family: Hawk Point
SMU BIOS Interface Version: 14
Version: v0.17.0 
PM Table Version: 4c0008
|        Name         |   Value   |     Parameter      |
|---------------------|-----------|--------------------|
| STAPM LIMIT         |    60.000 | stapm-limit        |
| STAPM VALUE         |     1.971 |                    |
| PPT LIMIT FAST      |    60.000 | fast-limit         |
| PPT VALUE FAST      |     2.176 |                    |
| PPT LIMIT SLOW      |    44.000 | slow-limit         |
| PPT VALUE SLOW      |     2.018 |                    |
| StapmTimeConst      |    46.076 | stapm-time         |
| SlowPPTTimeConst    |    90.609 | slow-time          |
| PPT LIMIT APU       |    45.000 | apu-slow-limit     |
| PPT VALUE APU       |       nan |                    |
| TDC LIMIT VDD       |    66.000 | vrm-current        |
| TDC VALUE VDD       |     0.691 |                    |
| TDC LIMIT SOC       |    18.000 | vrmsoc-current     |
| TDC VALUE SOC       |     1.177 |                    |
| EDC LIMIT VDD       |   130.000 | vrmmax-current     |
| EDC VALUE VDD       |    27.494 |                    |
| EDC LIMIT SOC       |    25.000 | vrmsocmax-current  |
| EDC VALUE SOC       |     1.684 |                    |
| THM LIMIT CORE      |   100.000 | tctl-temp          |
| THM VALUE CORE      |    39.545 |                    |
| STT LIMIT APU       |    38.000 | apu-skin-temp      |
| STT VALUE APU       |    31.090 |                    |
| STT LIMIT dGPU      |     0.000 | dgpu-skin-temp     |
| STT VALUE dGPU      |     0.000 |                    |
| CCLK Boost SETPOINT |       nan | power-saving /     |
| CCLK BUSY VALUE     |       nan | max-performance    |

xiota, 2025-06-13:
IMO, someone should flag AUR package.  Comments from last month asking to switch fork were ignored.