FlyGoat/RyzenAdj #355: get_core_* stopped working

ogoshen, 2025-06-09:
Getting nonsensical values.
Works on 0.16.0, issue started at 7d1c2c4da35369d3db9475ef17f97d52fa5ddbab

ogoshen, 2025-08-05:
Apologies my initial issue wan't very helpful or informatives.
Pretty much on the nose though, issue is the arithmetic expression in the `_read_float_value` macro expansion.

Fix could either go at call sites (but there are 4 more commits like this):

```diff
diff --git a/lib/api.c b/lib/api.c
index a4db018..583e2dc 100644
--- a/lib/api.c
+++ b/lib/api.c
@@ -2102,8 +2102,8 @@ EXP float CALL get_core_power(ryzen_access ry, uint32_t core) {
                default:
                        return NAN;
        }
-       
-       _read_float_value(baseOffset + (core * 4));
+
+       _read_float_value((baseOffset + (core * 4)));
 }
 
 EXP float CALL get_core_volt(ryzen_access ry, uint32_t core) {
```

Or in the macro body:

```diff
diff --git a/lib/api.c b/lib/api.c
index a4db018..d46d80a 100644
--- a/lib/api.c
+++ b/lib/api.c
@@ -427,7 +427,7 @@ do {                            \
 do {                                               \
 	if(!ry->table_values)                           \
 		return NAN;                                 \
-	return ry->table_values[OFFSET / 4];            \
+	return ry->table_values[(OFFSET) / 4];          \
 } while (0);
``` 

Bisection from 0.16.0 lead to this: 

dc2f34c
```
<CDLL './libryzenadj.so.dc2f34c194e400685c97510c1eb6cbe66bb8cfff', handle 6023d412bf70 at 0x7836702c5940>
bios_if_ver 18
table_ver 0x400005
stapm_limit 88.00000762939453
0 0.0
1 5.407528877258301
2 1.1142016649246216
3 4.8386969566345215
4 0.0
5 0.016400933265686035
6 0.6415473222732544
7 0.7246849536895752
```

7d1c2c4
```
<CDLL './libryzenadj.so.7d1c2c4da35369d3db9475ef17f97d52fa5ddbab', handle 5d412a9a3f70 at 0x7084accc5940>
bios_if_ver 18
table_ver 0x400005
stapm_limit 88.00000762939453
0 7.0375670973874695e+28
1 7.155541115427873e+22
2 1.2429487989279552e+22
3 3.203902637985423e-12
4 4.7417842493842954e+30
5 0.05966322496533394
6 220525664.0
7 13.958069801330566
```

Values above are wrong.

I looked at the table data itself and it looked fine, finally narrowed it down to the macro.

I saw that `init_table` did help things, data was usually more zeroed. 
But when the offset is correct it makes no difference.

Script used:

```py
import os, sys, time
from ctypes import *
from shutil import copyfile

lib_path = os.path.dirname(os.path.abspath(__file__))
os.chdir(lib_path)

if sys.platform == 'win32' or sys.platform == 'cygwin':
    try:
        os.add_dll_directory(lib_path)
    except AttributeError:
        pass #not needed for old python version

    winring0_driver_file_path = os.path.join(os.path.dirname(os.path.abspath(sys.executable)), 'WinRing0x64.sys')
    if not os.path.isfile(winring0_driver_file_path):
        copyfile(os.path.join(lib_path, 'WinRing0x64.sys'), winring0_driver_file_path)

    lib = cdll.LoadLibrary('libryzenadj')
else:
    lib = cdll.LoadLibrary('./libryzenadj.so')

print(lib)

# define ctype mappings for types which can not be mapped automatically
lib.init_ryzenadj.restype = c_void_p
lib.init_table.argtypes = [c_void_p]
lib.refresh_table.argtypes = [c_void_p]
lib.get_core_power.restype = c_float
lib.get_core_power.argtypes = [c_void_p, c_uint32]
lib.get_bios_if_ver.argtypes = [c_void_p]
lib.get_bios_if_ver.restype = c_int
lib.get_table_ver.argtypes = [c_void_p]
lib.get_table_ver.restype = c_int
lib.get_stapm_limit.argtypes = [c_void_p]
lib.get_stapm_limit.restype = c_float
lib.init_table.argtypes = [c_void_p]
lib.init_table.restype = c_int

ry = lib.init_ryzenadj()

if not ry:
    sys.exit("RyzenAdj could not get initialized")

print("bios_if_ver", lib.get_bios_if_ver(ry))
print("table_ver", hex(lib.get_table_ver(ry)))

print('stapm_limit', lib.get_stapm_limit(ry))

# print(lib.init_table(ry))

lib.refresh_table(ry)
for i in range(8):
    print(i, lib.get_core_power(ry, i))

```

ogoshen, 2025-08-07:
Sure thing, I'll open a PR.

Got a couple of others here as well that I've been meaning to open, part of the frontend I hope to publish soon:
https://github.com/ogoshen/AppRyzen/blob/main/ryzenadj.patch

Falcosc, 2025-08-08:
Yes, please add the brackets in the macro instead in the callers to avoid this issue in the future
Having the fix in the macro makes it more clear why brackets are needed at all, since most of us are not used to working with inline code blocks via macros.
If you find more macros with division or multiplication, it would be nice to have brackets there as well.
Thank you