FlyGoat/RyzenAdj #356: Strix Point : Unable to get os_access Obj, check permission Unable to init ryzenadj

bigcaesar8, 2025-06-15:
I have compiled latest ryzen_smu amkillam according the instructions in the main page. Then dkms installed
`git clone https://github.com/amkillam/ryzen_smu # Active fork of the original module
(cd ryzen_smu/ && sudo make dkms-install)`

Secure boot is disabled.
then rebooted
dmesg seems correct
`$ sudo dmesg |grep ryzen
[   10.837869] ryzen_smu: loading out-of-tree module taints kernel.
[   10.837882] ryzen_smu: module verification failed: signature and/or required key missing - tainting kernel
[   10.844182] ryzen_smu: loading version: 0.1.7
[   10.844190] ryzen_smu: CPUID: family 0x1A, model 0x24, stepping 0x0, package 0x1
[   10.844193] ryzen_smu: Family Codename: Strix Point
[   10.844494] ryzen_smu: SMU v11.93.4.0
[   10.845915] ryzen_smu: Unknown PM table version: 0x005D0009
[   10.845922] ryzen_smu: Failed to probe the PM table -- disabling feature (249)
`

Module is loaded
`$ cat /proc/modules | grep ryzen
ryzen_smu 45056 0 - Live 0x0000000000000000 (OE)
`
iomem=relaxed was applied to kernel

I then built Ryzenadj according to the following instructions on main page
`git clone https://github.com/FlyGoat/RyzenAdj
cd RyzenAdj
cmake -B build -DCMAKE_BUILD_TYPE=Release
make -C build -j"$(nproc)"
sudo cp -v build/ryzenadj /usr/local/bin/`

BUT when running ryzen adj for the first time I get to following error:
`$ sudo ryzenadj -i
detected compatible ryzen_smu kernel module
Unable to get os_access Obj, check permission
Unable to init ryzenadj
`
I have searched for this error but cannot see it anywhere. I know strix support is very new. And I would really like to 'undervolt' with set coall.

Any help would be appreciated.

vhqtvn, 2025-06-15:
Add `iomem=relaxed` to your boot args.

bigcaesar8, 2025-06-15:
Please see above. It's already been applied to kernel in grub parameters. Still have this issue.

vhqtvn, 2025-06-15:
ryzen_smu doesnt support 0x005D0009 yet, just use only ryzenadj is enough to read some values. I'm using 0.17.0.

bigcaesar8, 2025-06-16:
cat /proc/cmdline shows its added.
Interesting I see that amkillam ryzen_smu claims to support strix point but just not that 5D0009 table?
When I remove ryzen_smu and run sudo ryzenadj --set-coall=-15
I get the following

```
no compatible ryzen_smu kernel module found, fallback to /dev/mem
set_coall is rejected by SMU
```
Does this mean I am out of luck until ryzen_smu gets support for my table?

bigcaesar8, 2025-06-16:
tl;dr looks like ryzen_smu doesnt support my table yet. and set-coall is undocumented.
I just uninstalled ryzen_smu and did set-coall right and seems to work fine now.

I sorted it:
Needed to do 0x100000 - 15 to get the uint32 number i need for -15. 
brought me to this 
$ sudo ryzenadj --set-coall=0xFFFF1
no compatible ryzen_smu kernel module found, fallback to /dev/mem
Sucessfully set coall to 1048561

This is the link with the simplest explanation of the undocumented u32 aspect of set-coall
https://github.com/aarron-lee/SimpleDeckyTDP/issues/58#issuecomment-2814065707

closed 2025-06-16 by bigcaesar8