FlyGoat/RyzenAdj #357: Missing `ryzen_smu` 0.1.7

queenkjuul, 2025-06-23:
I followed the instructions in the README to install `ryzen_smu` (clone https://github.com/amkillam/ryzen_smu.git and install the module)

Building off of that repo's main branch gets me a module version of `0.1.6` and RyzenAdj requires `0.1.7` - there are no branches, tags, or releases in that repo, so I'm not sure where 0.1.7 lives?

talpadk, 2025-07-27:
I have the same issue, I dug a bit deeper.

The problem is caused by lib/linux/osdep_linux.c is_ryzen_smu_driver_compatible() which tries to open /sys/kernel/ryzen_smu_drv/drv_version to determine the version of the module.

However the entire folder /sys/kernel/ryzen_smu_drv does not exist. 
I have found the version information elsewhere /sys/module/ryzen_smu/version, but there are other files also missing the I could not find a replacement for.

I have been able to compile ryzen_smu and RyzenAdj with success on a 4750U based laptop with a stock Debian unstable kernel.

However here I'm compiling on a "ai max+ 395" and needs/uses a custom kernel 6.15.8 with a config based on the latest Debian unstable kernel.

talpadk, 2025-07-27:
Which for me is caused by ryzen_smu not supporting my CPU yet

from smu.c

  // Zen 5
  else if (cpu_family == 0x1a) {
    switch (cpu_model) {
    case 0x24:
      g_smu.codename = CODENAME_STRIXPOINT;
      break;
    case 0x44:
      g_smu.codename = CODENAME_GRANITERIDGE;
      break;
    case 0x70: // Strix Halo (AI MAX+ 395)
    default:
      pr_err("CPUID: Unknown Zen5/6 processor model: 0x%X (CPUID: 0x%08X)",
             cpu_model, cpuid);
      return -2;
    }