FlyGoat/RyzenAdj #363: Unable to --info or --dump-table on a AMD Ryzen AI Max+ 395

talpadk, 2025-08-09:
Using

RyzenAdj
branch master
sha 7aeb2f4869ee52ac161ee4cb4871e29113487885

ryzen_smu
https://github.com/CkNoSFeRaTU/ryzen_smu
branch dev
sha  23680fdd5e6806b7f3fe645771ac0e508c029dd9

On a
System: ASUSTeK product: ROG Flow Z13 GZ302EA
CPU: 16-core AMD RYZEN AI MAX+ 395 w/ Radeon 8060S (-MT MCP-)

 I get the following error messages
**sudo ./ryzenadj --info** 
```
detected compatible ryzen_smu kernel module
CPU Family: Strix Halo
SMU BIOS Interface Version: 25
Version: v0.17.0 
refresh_table failed
Unable to init power metric table: -5, this does not affect adjustments because it is only needed for monitoring.
```

**sudo ./ryzenadj --dump-table**
```
detected compatible ryzen_smu kernel module
refresh_table failed
Unable to init power metric table: -5, this does not affect adjustments because it is only needed for monitoring.
```

Running the script 
https://gist.github.com/CkNoSFeRaTU/d40ffc6c82c14ae205838bf98f87aeb3
does dump table data

talpadk, 2025-08-09:
Issue #350 seems to be related

kayti, 2025-10-21:
I have the same issue with Z2 extreme in Legion Go 2 (supposedly cutted version of 370).