FlyGoat/RyzenAdj #365: Unable to get RSMU Obj

winstonma, 2025-08-14:
I just checkout the https://github.com/FlyGoat/RyzenAdj/commit/377a5a942e867c3745eec3919558f79061ae0b1e commit, compile and run `ryzenadj`. And then I got the following error:

```bash
$ sudo build/ryzenadj -i
no compatible ryzen_smu kernel module found, fallback to /dev/mem
Unable to get RSMU Obj
Unable to init ryzenadj
```

Falling back to https://github.com/FlyGoat/RyzenAdj/commit/7aeb2f4869ee52ac161ee4cb4871e29113487885 and ryzenadj is normal.

SzabolcsGergely, 2025-08-16:
Same, AI max 395+

kstoykov, 2025-08-17:
Same, Ryzen HX AI 370

axze-az, 2025-08-17:
Same, Ryzen 7 5700g

g4bwy, 2025-08-20:
> Originating from this PR [#359](https://github.com/FlyGoat/RyzenAdj/pull/359) , a `break;` was missing. A user also reported this error before the PR was merged.

Hi, it seems more than one break was missing...

after fixing the missing break in ryzen_nb_smu_ops.c, this commit completely breaks (no pun intended) support on my Cezanne CPU, eg. in set_stapm_limit here https://github.com/FlyGoat/RyzenAdj/blob/master/lib/api.c#L458 the switch/case falls through DRAGONRANGE/FIRERANGE case unconditionnally.

the result is SMU hanging whenever I try to adjust the limits, with some random CPU resets

honjow, 2025-08-20:
> > Originating from this PR [#359](https://github.com/FlyGoat/RyzenAdj/pull/359) , a `break;` was missing. A user also reported this error before the PR was merged.
> 
> Hi, it seems more than one break was missing...
> 
> after fixing the missing break in ryzen_nb_smu_ops.c, this commit completely breaks (no pun intended) support on my Cezanne CPU, eg. in set_stapm_limit here https://github.com/FlyGoat/RyzenAdj/blob/master/lib/api.c#L458 the switch/case falls through DRAGONRANGE/FIRERANGE case unconditionnally (the same issue appears in most set_* functions, because there was no break before the default fall-through to begin with in most cases).
> 
> the result is SMU hanging whenever I try to adjust the limits, with some random CPU resets

so sad 😂

FlyGoat, 2025-08-20:
Just pushed a fix commit, I finally got another AMD system to test :-)

Sorry for my oversight.

closed 2025-08-20 by winstonma